mongoose-os-libs/i2c #14: The MPU-6050 is not detected (ESP8266 + GY521 sensor)

fbalestrini, 2020-05-05:
The MPU-6050 is not detected using an ESP8266.

**Hardware**

- WeMos D1 mini board (ESP8266)
- GY521 sensor (MPU-6050)

**Connections details**
|GY-521 pin|WeMos board pin|
|--|--|
|VCC|3V3|
|GND|GND|
|SCL|D1 (GPIO 5)|
|SCL|D1 (GPIO 5)|
|AD0|3V3|

**The mos.yaml**
```yaml
author:
description:
version: 1.0
manifest_version: 2017-05-18
libs_version: ${mos.version}
modules_version: ${mos.version}
mongoose_os_version: ${mos.version}

config_schema:
  - ["i2c.enable", true]
  - ["i2c.sda_gpio", 4]
  - ["i2c.scl_gpio", 5]

sources:
  - src
filesystem:
  - fs

libs:
  - origin: https://github.com/mongoose-os-libs/boards
  - origin: https://github.com/mongoose-os-libs/i2c
```
**The main.c**
```C
#include "mgos.h"
#include "mgos_i2c.h"

enum mgos_app_init_result mgos_app_init(void) {
  LOG(LL_INFO, ("Scanning I2C bus for devices"));
  struct mgos_i2c *conn = mgos_i2c_get_global();
  for (int i = 0; i < 0x78; i++) {
    if (mgos_i2c_write(conn, i, NULL, 0, true)) {
      LOG(LL_INFO, ("Found device at I2C address 0x%02x", i));
    }
  }
  return MGOS_APP_INIT_SUCCESS;
}
```
**The output**
```console
...
[May  5 16:09:54.515] mgos_i2c_gpio_maste:250 I2C GPIO init ok (SDA: 4, SCL: 5, freq: 100000)
...
[May  5 16:09:55.718] main.c:5               Scanning I2C bus for devices
[May  5 16:10:00.718] main.c:5               Scanning I2C bus for devices
[May  5 16:10:05.719] main.c:5               Scanning I2C bus for devices
...
```

fbalestrini, 2020-05-05:
@pimvanpelt I saw your great job checking the [imu-demo](https://github.com/mongoose-os-apps/imu-demo). I kindly ask your support on fixing the issue I am facing.

pimvanpelt, 2020-05-05:
The I2C driver in Mongoose OS works; it's likely that you have a wiring issue, or a termination / bus loading issue. Unfortunately, I cannot help you with wiring and chip detection issues.

You can try to scope the signal lines to ensure they are working. You can measure to make sure they are pulled high (10kOhm is typical on SDA and SCL lines). You can try a different ESP8266, and a different GY512 sensor, and ultimately a different implementation (like Arduino or PlatformIO).

fbalestrini, 2020-05-06:
Before opening this issue, I uploaded an Arduino sketch (using the Wire library) and everthing was working fine. I was able to detect and read sensor's registers without any problem. Same hardware, same wiring schema. Nothing was changed apart the firmaware.

**FIXED**: however, I pulled up the bus using `4.7kOhm` resistors according your great suggestions, and now everything is working fine using Mongoose OS as well. I thank you very much.

closed 2020-05-06 by fbalestrini