mongoose-os-libs/i2c #16: error: Array 's_buses[1]' accessed at index 2, which is out of bounds.

Harvie, 2020-12-05:
I just ran static analysis by cppcheck and it found this in i2c master code:

```
$ cppcheck i2c/
Checking i2c/src/cc32xx/cc32xx_i2c_master.c ...
Checking i2c/src/cc32xx/cc32xx_i2c_master.c: CC3200_I2C_DEBUG...
1/5 files checked 19% done
Checking i2c/src/esp32/esp32_i2c_master.c ...
2/5 files checked 53% done
Checking i2c/src/gpio/mgos_i2c_gpio_master.c ...
3/5 files checked 72% done
Checking i2c/src/mgos_i2c_master.c ...
Checking i2c/src/mgos_i2c_master.c: MGOS_SYS_CONFIG_HAVE_I2C1...
Checking i2c/src/mgos_i2c_master.c: MGOS_SYS_CONFIG_HAVE_I2C1;MGOS_SYS_CONFIG_HAVE_I2C2...
Checking i2c/src/mgos_i2c_master.c: MGOS_SYS_CONFIG_HAVE_I2C2...
i2c/src/mgos_i2c_master.c:95:12: error: Array 's_buses[1]' accessed at index 2, which is out of bounds. [arrayIndexOutOfBounds]
    s_buses[2] = mgos_i2c_create(mgos_sys_config_get_i2c2());
           ^
i2c/src/mgos_i2c_master.c:96:16: error: Array 's_buses[1]' accessed at index 2, which is out of bounds. [arrayIndexOutOfBounds]
    if (s_buses[2] == NULL) return false;
               ^
4/5 files checked 88% done
Checking i2c/src/ubuntu/ubuntu_i2c_master.c ...
5/5 files checked 100% done
```

rojer, 2020-12-05:
no, i believe this is false alarm.

closed 2020-12-05 by rojer

Harvie, 2020-12-05:
yeah sorry, now i see it is handled by ifdefs...